FujiNetWIFI/fujinet-hardware #2: Audio interference if FujiNet OFF and Atari ON, ESP32 getting backpowered

mozzwald, 2020-09-13:
If FujiNet is turned OFF and connected to a powered up Atari, the GPIOs on the esp32 can be reverse biased and partially power up the board. This is noticeable as a buzzing or ticking noise which appears thru the Atari SIO AUDIO_IN and then out to speakers/TV.  Basically, if the fujinet is turned OFF and Atari is ON, any SIO pin that goes high will back feed power to the ESP32 (I measure ~1-1.5V on VCC).

I am looking at using a bus switch IC to cut off the inputs when FujiNet is powered OFF.

billkendrick, 2020-10-18:
With the FujiNet off, and USB power in, I get both the ticking (it seems faster?), and a ton of static noise, if it's part of a larger chain of devices.  In my case:

- 1200XL
- #FujiNet (off, but USB power connected)
- 1050 floppy drive
- 850 serial interface
- SIO2SD v1.3

(With the USB power unplugged, I get just the ticking.)

mozzwald, 2020-12-25:
Fixed with 47ee29afcc630a1f6788ae41a77ae9899965f131

closed 2020-12-25 by mozzwald