FujiNetWIFI/fujinet-hardware #6: Unable to print power switch using FDM printer

sebobialy, 2021-01-15:
It is impossible to properly slice and print the power switch due to too thin walls. There are so thin, that Cura slicer throws them away, leaving the switch unprintable.
[image]
Also, the switch is very fragile to be used in standard printer settings like 0.6mm nozzles. Please consider redesigning this piece to make it more robust. Both walls and base should be thicker.

mozzwald, 2021-01-15:
> It is impossible to properly slice and print the power switch due to too thin walls. There are so thin, that Cura slicer throws them away, leaving the switch unprintable.

I am able to slice and print this with a 0.4mm nozzle and line width set to 0.4mm. I do need to print multiple parts at a time (5 or more should work) otherwise the filament does not have enough time to cool. My results are good. 

The design is open if you would like to modify it to work with your printer.

sebobialy, 2021-01-15:
It is not about printing problems, but slicing such a thin model. Cura is still not happy with 0.4mm line width:
[image]
Is there any source for this case other than stl files? Version 1.0 had ".step" file, now it is gone from trunk.

TheRobotFactory, 2021-01-15:
I was able to get around this bug by ticking "Print thin walls" is the shell settings, it still prints it as helper rather than wall but it was the only way I could get it to slice at 0.4mm nozzle. Model does need to be slightly thicker. 
[image]

In Prusaslicer it was the same unless I enable 'detect thin walls' but prusaslicer does a better job of incorporating the thin walls back into the model.

[image]

mozzwald, 2021-01-15:
I've reopened the issue if someone would like to make a fix. 

I will add step files to the repo soon. Until then you can get them from tinkercad using the links in the readme

mozzwald, 2021-01-16:
> Is there any source for this case other than stl files? Version 1.0 had ".step" file, now it is gone from trunk in 1.3.

Looking deeper, the only STEP file I posted for v1.0 was for a 3D file of the PCB which I can add for v1.3. The case designs, however, are only exportable as STL, OBJ and GLTF (.GLB). If you open the design in Tinkercad you can _Send To_ Fusion 360 if you use that software.

iCityWork, 2021-01-17:
It is working for me with Prusa, yes you need to turn on detect thin walls.  
[image]

mozzwald, 2021-12-03:
Closing due to inactivity

closed 2021-12-03 by mozzwald