FujiNetWIFI/fujinet-hardware #7: SIO Receptacle pins are short preventing some cables from remaining seated

mozzwald, 2021-03-14:
Some users have reported that their SIO cables pop out or fall out of the FujiNet receptacle. The pins used on FujiNet are shorter than an original SIO receptacle. To combat this issue I am testing some different pins that are longer. Preliminary testing looks promising.
[image]
[image]
[image]

iCityWork, 2021-12-03:
Do you have the 3D STL for the little pin jig you made to aid in soldering these new pins in the correct place?
Thanks,
Gavin

mozzwald, 2021-12-03:
> Do you have the 3D STL for the little pin jig you made to aid in soldering these new pins in the correct place? Thanks, Gavin

I do and will post them this weekend, thx for the reminder ;)

Closing, issue fixed with d48371e

closed 2021-12-03 by mozzwald

iCityWork, 2021-12-08:
call me crazy but I am not seeing the STL for it yet. What am I missing?

mozzwald, 2021-12-08:
> call me crazy but I am not seeing the STL for it yet. What am I missing?

I tagged you on AA with the file:

https://atariage.com/forums/topic/327694-fujinet-v16-hardware/?do=findComment&comment=4959920

When I have time, I will put them here with some instructions

iCityWork, 2021-12-09:
Oh thank you sir, I don't go to AA that often.  Much appreciated.