FujiNetWIFI/fujinet-hardware #9: USB port possibly prone to breaking off on 800

sprive, 2021-10-04:
Hey all

Just end user board feedback (v1.0 rev)

First -- thanks! :-) 

Second - on a 400/800 the FN USB port is always needed. I suspect this adds more stress (always in) than plugging and unplugging.

My suggestion probably adds too much cost (but it's what I will try, in attempted repair):
* Maybe a panel-mount connector
* or through-hole with screw-in wings/tabs like the A121025TR.

Basically... something gives the connector someplace to offload stress. 
cheers

sprive, 2021-10-18:
Thanks - you've given me a better place to attempt a panel-mount cable.

As you say - bumping the cable while plugged in can damage the port.... but this is not something that will be changed. 

I hear that.... still, please consider the problem further (if another cause for revision presents itself). Holes to either side would allow one to reinforce with a zip tie. Thanks again.

rcamp5656, 2021-11-29:
does not the case have a slot for putting a cable into the Fujinet, or at least mine does, the case takes all of the stress off of the connector if done right

sprive, 2021-11-30:
> does not the case have a slot for putting a cable into the Fujinet, or at least mine does, the case takes all of the stress off of the connector if done right

Ah, a different perspective. TY. "yes and no" to your question.

Basically the case opening is an oversized hole the cable goes into. It is not at all a "close" fit, and you would need a close (friction) fit around the USB shroud, so the case might potentially absorb force. 

3D printers generally have trouble with case walls less than 2mm, and the USB shroud protrudes from the PCB only 1mm.  Someone might still be able to design a case with a tight fit around the port (inside reinforcement) but the current "hole in the wall" wouldn't work.

I just received DC barrel jacks, threaded kind with a nut. Then for me the USB is only used for flashing updates.

Alternatively, a FN on an 800 doesn't need external power if one buys the Incognito board. Incognito adds a tiny boot delay which means the FN is initialized by the time the OS checks SIO for boot drive.