FujiNetWIFI/fujinet-hardware #12: Beta 2 firmware broken

chuckcaley, 2022-11-25:
I have been installing the beta 2 firmware with no problems, but as of yesterday i installed it on 2 of my fujinets and it has killed the sd card. The sd card was 16gb but now it says 512kb and wont even format. Also it is hit and miss as to weather it wants to connect to my router.

mozzwald, 2022-11-25:
It is possible your SD cards were fake and only reporting that they were 16GB. I recommend you try formatting the SD card as FAT32 with https://www.sdcard.org/downloads/formatter/

Please capture debug output of when a problem is happening so we can try to figure out what the cause is. https://github.com/FujiNetWIFI/fujinet-platformio/wiki/FujiNet-Flasher#capturing-serial-debug-output

mozzwald, 2022-11-25:
> The cards should be fine. The first card was used in my fujinet which had the same firmware, but from the other week. Which worked fine.

Please try using the SD Card formatter linked preivously to get the cards in working order on your computer before trying them in the fujinet again.

> Also when you pressed option 1 or 2 it would come up with the reconnect screen option but it would start from the middle of the screen an wrap round. I will reflash and post a picture on Monday as I am away until then.

The screen glitch is a known issue and is due to a bug in the CONFIG program adding an extra line to the display list.

mozzwald, 2023-04-27:
Closing due to no activity

closed 2023-04-27 by mozzwald