FujiNetWIFI/fujinet-hardware #14: ATARI FujiNet FN32ROV-1.6 BOM / Trouble Sourcing ESD5Z5.0T1G Protection Diode / Replacement?

VonHirsch, 2023-04-27:
Hello, I had some 1.6 PCB's created awhile ago and I'm preparing to buy some components (see link to Octopart BOM below)

I'm having trouble sourcing the ESD5Z5.0T1G protection diode, would you happen to know a replacement part?  Thanks!

Link to BOM:
https://octopart.com/bom-tool/wwdTgpTK

mozzwald, 2023-04-27:
This is an equivalent part, currently available from Digikey: ESD5V0D5-TP

closed 2023-04-27 by mozzwald

VonHirsch, 2023-04-27:
TY!

VonHirsch, 2023-04-27:
@mozzwald also if you don't mind, do you think [this ESP32 part ](https://www.mouser.com/ProductDetail/Espressif-Systems/ESP32-WROVER-E-N16R8?qs=Li%252BoUPsLEnsfUZ%252By2eNF6g%3D%3D)will be compatible with the build?

mozzwald, 2023-04-27:
> @mozzwald also if you don't mind, do you think [this ESP32 part ](https://www.mouser.com/ProductDetail/Espressif-Systems/ESP32-WROVER-E-N16R8?qs=Li%252BoUPsLEnsfUZ%252By2eNF6g%3D%3D)will be compatible with the build?

That is the correct module

VonHirsch, 2023-04-27:
@mozzwald outstanding, tysm!