FujiNetWIFI/fujinet-hardware #15: Problem with "FujiNet XL v1.7 Case Front" STL file (multiple shells)

aleksander-martin, 2023-08-23:
I tried to order all case elements at JLCPCB. Unfortunately they wrote me, that there is a little problem with this part of case. They wrote, that "FujiNet XE v1.7 Case Front.stl" file consists of separate shells and the whole element should be as one shell. They asked me to combine all the separate shells into one with connections. These are the two mounts at the hole for the SIO connector. I have no source files and it's impossible to do this from STL file. Could you fix it please?

mozzwald, 2023-08-24:
Thanks for letting us know. I've updated the XE front case in commit 13eca1497a4e363a38683fdc3f2b80978d17f560. These were designed with Tinkercad and link to the "source" is available in the README file.

closed 2023-08-24 by mozzwald

dlxmax, 2024-04-11:
@mozzwald I just encountered the same problem, but with "FujiNet 400 800 v1.7 Case Front.stl"
Thanks