FujiNetWIFI/fujinet-hardware #21: Problem with "FujiNet 400 800 v1.7 Case Front" STL file (multiple shells)

dlxmax, 2024-04-12:
I tried to order all case elements at JLCPCB. Unfortunately, they said there is a problem with part of case. The "FujiNet 400 800 v1.7 Case Front.stl" file consists of separate shells and the whole element should be as one shell. They asked me to combine all the separate shells into one with connections. These are the two mounts at the hole for the SIO connector. It's beyond my ability to do this with the STL file. Could you fix it please?

[image]

retrobrian, 2024-04-12:
This is strange. My tests show that the stl file is manifold and water tight. So it should be okay. I wonder if JLCPCB is seeing something corrupt in the file.

[image]

I re-saved the file and attached it to this comment. I uploaded the file to JLCPCB's quote service, and it accepted the file, but I'm thinking JLCPCB contacted you after you submitted the order?

Would you like to try and send them my uploaded file?

[FujiNet 400 800 v1.7 Case Front-test.zip](https://github.com/FujiNetWIFI/fujinet-hardware/files/14953023/FujiNet.400.800.v1.7.Case.Front-test.zip)

dlxmax, 2024-04-12:
Thank you for the file, but it gave them the same problem (see image).
Interestingly, the same issue (https://github.com/FujiNetWIFI/fujinet-hardware/issues/15) also affected the XE case, but was resolved by @mozzwald 

[image]

retrobrian, 2024-04-12:
Thank you for pointing out the previous issue. I'd still like to help you move forward with your print.

I loaded up "FujiNet 400 800 v1.7 Case" in Tinkercad, grouped the two standoffs with the case, and exported a new STL.

The new STL is attached to this comment.

BTW, I was able to see the problem in the original STL by using Meshmixer. I also checked the new STL I created and it is showing as one shell now.

[Copy of FujiNet XL v1.7 Case-grouped.zip](https://github.com/FujiNetWIFI/fujinet-hardware/files/14960882/Copy.of.FujiNet.XL.v1.7.Case-grouped.zip)

trevorbattle, 2024-04-12:
[image]

Not sure if helpful but Prusaslider always warns about the rear for me as well. It quietly fixes it though.

mozzwald, 2024-04-15:
> I loaded up "FujiNet 400 800 v1.7 Case" in Tinkercad, grouped the two standoffs with the case, and exported a new STL.

this is the way, but it looks like you did the XL case, not the 400/800 case

> They asked me to combine all the separate shells into one with connections

My slicer does not complain about this and happily combines them automatically. Please try the attached file and if it works I will update the repo with it.

[FujiNet 400 800 v1.7 Case Front.stl.zip](https://github.com/FujiNetWIFI/fujinet-hardware/files/14984780/FujiNet.400.800.v1.7.Case.Front.stl.zip)

dlxmax, 2024-04-16:
@mozzwald Thank you for catching that wrong case!
However, the file you sent still has the same problem as before--3 bodies.
Perhaps you sent the wrong file? I appreciate your help.

dlxmax, 2024-04-16:
@mozzwald thanks for the tip about the repair service. I'm in the queue, though it looks to be a several hour wait.

Regarding the STL you sent, I imported it as a mesh in Fusion, converted the mesh to solid body--and got the 3 piece result you see in this image. I even downloaded your link twice just to make sure that I didn't mix it up with an old version. Unfortunately, I cannot get Fusion to merge the bodies together, as I am a long-term novice at CAD.

[image]

dlxmax, 2024-04-16:
@mozzwald I tried the online fixer you suggested, but it didn't fix it.
```
-> In queue. Waiting before you in the queue: 0 (refreshing each second)
-> Reading file and indexing vertices
-> Analysed your file:
--> 0 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
--> 0 Non-manifold edges (?)
--> 0 Inverted faces (?)
--> 0 Degenerate faces (?)
--> 0 Duplicate faces (?)
--> 0 Disjoint shells (?)
-> Repairing: 100.00%
----- Repair completed in 12530ms ------
-> Vertex count changed from 2837 to 3242 (+405)
-> Triangle count changed from 5694 to 6504 (+810)
-> FujiNet 400 800 v1.7 Case Front.stl available for download. Click the button below to download
```
[image]

dlxmax, 2024-04-18:
@mozzwald Do you have any other ideas?
The fix you did for issue https://github.com/FujiNetWIFI/fujinet-hardware/issues/15 was perfect.

mozzwald, 2024-04-18:
@dlxmax try this and let me know if it works
[400-800-stl-test.zip](https://github.com/FujiNetWIFI/fujinet-hardware/files/15030829/400-800-stl-test.zip)

dlxmax, 2024-04-21:
@mozzwald THANK YOU! It passed my tests and JLC's too.
I noticed you also included v1.7 of the case rear, while the repository only has v1.6 for all types, so I also went ahead and updated my order with it too.